fact-project/fact-tools #92: JSONWriter does not provide valid JSON

dneise, 2016-02-01:
We all know, that the JSONWriter does not create valid JSON files, unless `writeListOfItems = "true"` is set.
This is an old story already.

But it also does not treat infinite values and not-a-number values correctly.

We encountered files, looking like this:

```json
{"some_key" : -Infinity,
"another_key": Infinity,
"still_another_one": NaN}
``` 

As you can see, this is not valid JSON. (We also checked it using http://jsonlint.com/)

---

We use `pandas.read_json` to parse these files, and found that by exchanging these values by some string representation of infinity and nan, the parser like it much better.

E.g.
```json
{"some_key" : "-inf",
"another_key": "inf",
"still_another_one": "nan"}
```

mackaiver, 2016-02-01:
I agree somewhat. However the reasoning behind writing these values is JS compatibility.  Despite what the linter says these value are common among JavaScript parsers in many browsers. 

See also [this google statement] (https://google-gson.googlecode.com/svn/trunk/gson/docs/javadocs/com/google/gson/GsonBuilder.html#serializeSpecialFloatingPointValues%28%29) about special floating point values in JSON. 

Maybe we can add another flag for Pandas compatible output.

dneise, 2016-02-01:
Under these circumstances, JSON disqualifies as a file format for scientic data storage and exchange.

Maybe the pandas community can be asked to support the same pseudo standard as google does...

We, however, should imho stop using JSON in order not to lead our colleagues into traps.

MaxNoe, 2016-02-02:
We use `json` or `jsonl` mainly as intermediate serialisation format. Most of us convert the multiple `json`-files that come from processing multiple runs to one big file in another format. E. g. `.root` or `hdf5`.

Note that `python`'s standard library `json` module understands the `JavaScript`/`Google` Pseudo-Standard:

```{python}
In [1]: import json
In [2]: json.loads('{"a": [5, Infinity]}')
Out[3]: {'a': [5, inf]}
```

So to get a json file using the `Infinity` and `NaN` into a pandas dataframe, you can do:

```{python}
import json

data = json.load('myfile.json')
df = pd.DataFrame(data)
```

dneise, 2016-02-02:
Even worse. Using something, which is obviously not useful for scientif

mackaiver, 2016-02-02:
See also Issue #73

closed 2016-02-09 by MaxNoe