fact-project/fact-tools #94: FITS writer

MaxNoe, 2016-02-02:
This would be useful for quite some situations.

MaxNoe, 2016-02-02:
http://nom-tam-fits.github.io/nom-tam-fits/intro.html

dneise, 2016-02-02:
this nom-tam part makes it sound funny ... why not just fits?

alexeyegorov, 2016-02-23:
I've made a pull request for writing FITS files using nom-tam-tam-tam FITS library. Tests are failing, but due to missing drs file. This happened first after I pulled newest changes from refactoring branch.

alexeyegorov, 2016-02-25:
There are some field, that we seem not use right now while reading FITS, such as PCOUNT or GCOUNT. They seem to be required normally? This would be important for the case if we would like to replace our FITS reader with nom-tam-fits reader.

MaxNoe, 2016-10-15:
This was added in #138

closed 2016-10-15 by MaxNoe