fact-project/fact-tools #95: Clean up

MaxNoe, 2016-02-04:
We have a huge number of branches.

Can the branch creators check if their branches are still needed and delete them, if this is not the case?

ftemme, 2016-02-10:
Branches which are only behind the master are now removed. There are a few branches which maybe are not in use anymore (last commit was 4 month or more ago):

- LightPulser by lena
- ringing by jens
- maxnoe/circularFit by max
- Corsika by Dominik B.
- triggerInvestigation by kai
- 2D_filters by relleums
- Snake by Dominik B.
 - this we should keep and at some time merge the important processors

Please can the corresponding authors comment, if the branches shall be kept?

MaxNoe, 2016-02-11:
The circularFit contains important stuff, I will rework it to match the current master shortly.

closed 2017-11-07 by MaxNoe