fact-project/fact-tools #100: I want to remove the RootASCIIWriter

mackaiver, 2016-02-05:
It has been deprecated for a while now. I'd like to remove it. At least during the coding sprint we have set up for next week.

mahnen, 2016-02-06:
I was using it for comparing FACT to MAGIC data. Would be nice for me to keep this function in.

MaxNoe, 2016-02-07:
If you just want to have a root file at the end of the day, you could consider using the `JSONWriter` like this:

```{xml}
<fact.io.JSONWriter url="file:test.json"  keys="key1,key2" specialDoubleValuesAsString="true" />
```

And use `pandas` and `root_pandas` like this:

```{python}
import pandas as pd
import root_pandas as rpd

data = pd.read_json('test.json')
data.to_root('test.root', 'treename')
```

MaxNoe, 2016-02-09:
@mahnen, @ftemme 

Could you describe your workflow with this processor? How do you convert the `.csv` to `.root`?

MaxNoe, 2016-02-09:
We discussed yesterday, that we want to keep this processor, and that it will be undeprecated if the todos in the first post are done.

mahnen, 2016-02-09:
Thanks, if you like I could still report on the workflow though. Exactly as you mentioned, I use the TTree.ReadFile:

`	if ( FactFilePaths.size() != FactRootFilePaths.size() ){
		// convert the txt files to root files
		for (std::string OneFactFile : FactFilePaths )
		{
			//generate a root tree file from the fact-tools output 
			//in order to read it in with the ROOT chain method later.
			std::string outbuf = OneFactFile;
			outbuf = outbuf.replace( outbuf.end()-3,outbuf.end(),"root");
			
			TFile f( outbuf.c_str() , "RECREATE");
			TTree fTree;
			fTree.ReadFile( OneFactFile.c_str() ,"",' '); 
			fTree.SetDirectory( &f );
			fTree.Write("",TObject::kOverwrite);
			f.Close();
			//std::cout << "file to be opened: " << OneFactFile << std::endl;
			//std::cout << "file to be written: " << outbuf << std::endl;
		}
	}`

MaxNoe, 2017-06-13:
I think nobody is using and anymore and with good reason.

closed 2017-06-13 by MaxNoe