fact-project/fact-tools #116: Better regressor for Disp

mahnen, 2016-03-01:
Hi guys,

I wrote a little something about the current status of the Disp-parameter. In particular it would be quite important to get this Parameter estimated better. 

[disp_rf.pdf](https://github.com/fact-project/fact-tools/files/152947/disp_rf.pdf)

mahnen, 2016-03-01:
small summary from the pdf:

- I to trained a disp RF and applied it to the mc with facttools
- Target for training: 'Distance' in MC of gammas equivalent to Kai's approach, use sklearn and then export to .pmml
- I used the 'apply model' processor in the jpmml branch.

Results:

- PSF got a bit better (~16%) → therefore bg now down to: ~ 75% of its value before, smaller tails. (probably much improvable with better disp-models)
- Problem of loss of gamma rays at highes energies after theta-cut: gone!

My opinion on it:

- The .pmml – apply model processor in facttools is super-useful.
We should definitely use it!

- You can check out the DispRF-xmls I used in the jpmml branch
of facttools. The regressor itself I uploaded to the Phido:

/fhgfs/groups/app/fact/simulated/DispRegressor_v1.pmml

- Imho this is the last issue in facttools which, when solved, would make this the
superior performing analysis in comparison to MARS.

MaxNoe, 2016-03-01:
Hey, 

Can you give more details on the plots you showed? I do not really understand what the different colors mean. Can you maybe add a plot where you just show the comparison for the old display method and your random forrest?

Also a correlation plot between the two methods might be interesting.

mahnen, 2016-03-21:
1) correlation plot between the two methods

[image]

2) comparison for the old disp method and your random forrest

[image]

mahnen, 2016-03-21:
I made some more plots. In theory, Disp = Distance for gamma-rays. The following plots are Disp vs Distance for the two methods. It looks to me rather like Danielas parametrization of Disp is a random number...

[image]
[image]

mahnen, 2016-03-21:
The last comparison is the size over theta^2 distribution. It shows that we can particularly gain something at higher sizes

[image]
[image]

ftemme, 2016-04-06:
Really nice, but I have some comments on that:

- I don't think that Disp = Distance, cause Distance doesn't take the angle to the source (alpha) into account. 
 - Maybe it is enough to use Disp = Distance / cos(alpha)
- In the MCs, Distance (and alpha) are already based on the reconstruction of the event (so extraction image cleaning ...) so I have difficulties to think about what is really the 'mc-truth' for disp.
- I have difficulties to think about the sign for the source reconstruction (so should disp be added in one or the other direction)
- Just as an information for all of you: The currently used disp parameterization is not the one from Daniela, but a very easy one only using width and length. Nice to see, that is has performance issues at high energies, so there is quite room for improvement left.

I would propose to have a brainstorming telcon (or skype con) next week (11.4. - 15.4.) to discuss this. But I agree with you, this is one of the main remaining issue FACT-Tools has to handle and your application of the RF regression is quite impressive.

mahnen, 2016-04-06:
I have looked up the definition of Dist in Sebastian Commichaus old PhD thesis. 
He there writes:

> The image parameters DIST and ALPHA are so-colled source-dependent image parameters, as they are calculated with respect to the source position given by the center of the camera.

The drawing goes as follows:
[image]

He further continues about the sign of the Disp parameter:

> Because of the head-tail ambiguity in the 2nd moment analysis, the Disp method provides two possible solutions for the location of the source along the mayor axis of the Hillas ellipse. A possibility to overcome the head-tail ambiguity is to make use of the asymmetry parameter M3LONG. ... The achieved ratio of correct head-tail assignment is in the order of 70%, which strongly depends on the considered energy range. 

The MC-truth (DIST) is really, like you mentioned, cleaning dependent etc..., which unfortunately breaks the streams-flow... you have to process it first and then build the Disp-RF and then run again. (or implement a RF training and application in facttools Java)?

MaxNoe, 2017-11-07:
Moved to our python classifier tools and works quite nicely

closed 2017-11-07 by MaxNoe