fact-project/fact-tools #117: RemoveSpikes possibly dangerous

dneise, 2016-03-10:
Here I show an example, where the `fact.datacorrection.RemoveSpikes` can largely distort the signal and thus lead to false results. This I regard as *possibly dangerous* although in the current std usecase of fact-tools it is **absolutely no problem**.

I use: 
 * [fact-tools-v0.13.0.jar](https://github.com/fact-project/fact-tools/releases/download/v0.13.0/fact-tools-v0.13.0.jar)
 * [viewer.xml](https://github.com/fact-project/fact-tools/blob/451336402df5eef6740c0ca6a61d9015dcf97e39/examples/viewer.xml)

---

I used the `viewer.xml` to look into `drs-time` runs. The drs-time calibration signal is a 250MHz periodic signal, visible in every 9th channel. 

This is the first event of `20140407_006.fits.fz` (calibrated with `20140407_004.drs.fits.gz`) chid 71 (2nd row from top, 3rd pixel from the right). 
[image]

As you can see the amplitude is not very stable and not every period of the signal creates a zero crossing. In fact hardly any period does produce a zero crossing. 

----

Same file, same event, same pixel, different zoom to the x-axis. The raw data looks like the amplitude should be pretty stable, but after the drs calibration, the amplitude is distorted.

[image]

----

When using only `viewer.xml` and the jar-file downoaded from the releases. I cannot comment out parts of the DRS calibration. I need to crawl deeply into the directory tree and find the calibration.xml in order to copy and modify it. Not very easy to accomplish imo.

However after getting my own copy of `calibration.xml` and commenting out `fact.datacorrection.RemoveSpikes` The same event looks like this:

[image]

This is pretty much, what I would have expected. 

----

MaxNoe, 2016-03-10:
Pretty successfull in removing those big spikes, isn't it?

ftemme, 2016-04-06:
As Max stated very laxly, the RemoveSpikes algorithm is doing exactly as it should. Unfortunately the upper part of the drs-time calibration signal has the same signature as the spikes which should be removed by the RemoveSpikes algorithm.

I am not sure how to handle this correctly. So as you already mentioned for the standard use cases everything works fine. At the moment I could only think of this drs-time runs, where the RemoveSpikes algorithm is a problem.

But in general sometimes people want to dig into the first steps of the analysis, so into what is done in calibration.xml. And therefore the way to accomplish this is, as you stated correctly, not very easy.
So I added a new minimal viewer (see 65c5910) in which only the calibration tasks are performed. Also it doesn't use the calibration.xml from the classpath, but the processors are specified directly in the xml file.
Also I changed the keys for the different steps of the processed data, so that they all can be shown in the viewer.

Hopefully this helps the next time someone wants to dig into raw, or just calibrated data. Unfortunately we have to be sure, that changes in calibration.xml will also be done in the minimal_viewer but I don't expect so many changes in the calibration in future.

I won't close this issue, because I am not sure if this really solve the issue.

dneise, 2016-04-06:
Wow!

Thanks for the long answer and the work. I didn't even think of providing something like a minimal viewer. Good idea.

I think an implementation, which does only remove spikes and not real signals would solve this issue.
Maybe it exists but we have no time to research it... maybe it does not exist.

So ... well .. maybe it can be put on hold?