fact-project/fact-tools #118: Cannot view files from 2012

dneise, 2016-03-10:
when doing this: (version 0.13.0) 

`ftv -Dinfile=file:20120123_119.fits.fz -Ddrsfile=file:20120123_117.drs.fits.gz`

`ftv` is just my alias for `java -jar <path to jar> <path to viewer.xml>

I get this:
~~~
Mar 10 2016 15:30:23 [ERROR] stream.io.SourceURL: Failed to register handler for protocol 'hdfs'
Mar 10 2016 15:30:23 [INFO ] stream.util.XIncluder:    url expanded to: 'file:/home/dneise/fact/ft/calibration.xml'
Mar 10 2016 15:30:23 [INFO ] stream.util.XIncluder:    url expanded to: 'classpath:/default/data/extraction.xml'
Mar 10 2016 15:30:23 [INFO ] stream.util.XIncluder:    url expanded to: 'classpath:/default/data/cleaning.xml'
Mar 10 2016 15:30:23 [WARN ] stream.runtime.setup.ParameterInjection: There are conflicting annotations for the field outputKey. Remove annotation from method setOutputKey.
Mar 10 2016 15:30:23 [INFO ] stream.runtime.ProcessContainer: ######## Sources ########
Mar 10 2016 15:30:23 [INFO ] stream.runtime.ProcessContainer: ########fact.io.zfits.ZFitsStream@59f95c5d########
Mar 10 2016 15:30:23 [INFO ] stream.runtime.ProcessContainer: 	==> DefaultProcess['2']
Mar 10 2016 15:30:23 [INFO ] stream.runtime.ProcessContainer: ######## RootSources ########
Mar 10 2016 15:30:23 [INFO ] stream.runtime.ProcessContainer: fact.io.zfits.ZFitsStream@59f95c5d
Mar 10 2016 15:30:23 [INFO ] stream.runtime.ProcessContainer: ######## NonRefSinks ########
Mar 10 2016 15:30:23 [INFO ] stream.runtime.ProcessContainer: ProcessContainer is initialized and ready to start:stream.runtime.ProcessContainer@5c0369c4
Mar 10 2016 15:30:23 [INFO ] stream.run: Starting process-container...
Mar 10 2016 15:30:23 [INFO ] fact.io.zfits.ZFitsStream: Read file: 20120123_119.fits.fz
Mar 10 2016 15:30:23 [INFO ] fact.io.zfits.ZFitsStream: Read file: 20120123_119.fits.fz
Mar 10 2016 15:30:23 [INFO ] fact.io.zfits.ZFitsStream: Found Table ZDrsCellOffsets
java.lang.NumberFormatException: For input string: "J      "
	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
	at java.lang.Integer.parseInt(Integer.java:580)
	at java.lang.Integer.parseInt(Integer.java:615)
	at fact.io.zfits.ZFitsTable.<init>(ZFitsTable.java:84)
	at fact.io.zfits.ZFitsUtil.skipToTable(ZFitsUtil.java:125)
	at fact.io.zfits.ZFitsStream.init(ZFitsStream.java:104)
	at stream.runtime.ProcessContainer.execute(ProcessContainer.java:633)
	at stream.runtime.ProcessContainer.run(ProcessContainer.java:594)
	at stream.run.main(run.java:174)
	at stream.run.main(run.java:147)
Exception in thread "main" java.lang.RuntimeException: java.lang.NumberFormatException: For input string: "J      "
	at stream.runtime.ProcessContainer.run(ProcessContainer.java:597)
	at stream.run.main(run.java:174)
	at stream.run.main(run.java:147)
Caused by: java.lang.NumberFormatException: For input string: "J      "
	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
	at java.lang.Integer.parseInt(Integer.java:580)
	at java.lang.Integer.parseInt(Integer.java:615)
	at fact.io.zfits.ZFitsTable.<init>(ZFitsTable.java:84)
	at fact.io.zfits.ZFitsUtil.skipToTable(ZFitsUtil.java:125)
	at fact.io.zfits.ZFitsStream.init(ZFitsStream.java:104)
	at stream.runtime.ProcessContainer.execute(ProcessContainer.java:633)
	at stream.runtime.ProcessContainer.run(ProcessContainer.java:594)
	... 2 more
~~~

In case you wonder:
Yes, I tried it with newer files. Then it works:
Eg:`ftv -Dinfile=file:20140407_006.fits.fz -Ddrsfile=file:20140407_004.drs.fits.gz`
works

mackaiver, 2016-03-19:
Hello,
I cannot find the specific file you referred to but it seems to work for `20120123_125.fits.fz` now.
Can you checkout out the branch [zfits_2012_fix](https://github.com/fact-project/fact-tools/tree/zfits_2012_fix) and try it out on  `20120123_119.fits.fz` ?

MaxNoe, 2016-04-06:
#123 was merged, this should solve this issue. Please reopen if you experience further problems

closed 2016-04-06 by MaxNoe