fact-project/fact-tools #120: filter: MovingMinimum

dneise, 2016-03-17:
For a trigger reconstruction, I think I need a a filter: MovingMinimum. Which works just like a Moving Average. 

In dec77775bd411b19ae8825e4e576a993937bc128 I added a first proposal. Should I already open a PR and you tell me in the discussion there, what is missing, e.g. tests and so on.

Or do you rather want to discuss it here? Or does it maybe already exist?

---

In this picture you can see what I use it for:

[image]

In light blueish you see 9 signals of a patch.
In gray the PatchAverage
In blue the "clipped" PatchAverage, i.e. the operation our 10ns delay cables do on the signal in our trigger branch (on the FTU). Implemented using a FirFilter.
In red the MovingMinimum of length 5. 

Here is a snipped from the XML

    <fact.statistics.PatchAverage key="DataCalibrated" outputKey="PA" />
    <fact.filter.FirFilter key="PA" outputKey="PA_cl" 
      coefficients="1., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., -0.6" />
    <fact.filter.MovingMinimum key="PA_cl" outputKey="PA_cl_min" length="5" /> 
    <fact.extraction.MaxAmplitude key="PA_cl_min" outputKey="PA_cl_min_MAX" />

By extracting the MaximumAmplitude of the the moving minima of the clipped patch average, I get something like: "The highest possible trigger threshold, which this patch would still have triggered."

MaxNoe, 2018-02-16:
Probably related to #263

MaxNoe, 2018-03-01:
This filter is in master since a long time

closed 2018-03-01 by MaxNoe