fact-project/fact-tools #134: something can distort pulses

dneise, 2016-05-19:
I was just clicking through a couple of events.

In particular: 20140104_139 event 546 (a muon ring)

Have a look at these two pixels, inside the ring. 
Their peaks are distorted, which most probably changes the integral.
The arrival time in this case is not affected.

I wonder, what processor might do this. Looks like jump correction.
Maybe this is unavoidable ... I have not further studied it ... just wanted to tell you, in case you didn't know something like this happened.

[image]
[image]

MaxNoe, 2016-05-19:
This really looks strange.

dneise, 2016-05-19:
I see the same also in other files from the same day, not only in this file ... not only in this event.