fact-project/fact-tools #137: regarding delays*.csv files

dneise, 2016-07-28:
The files:

`classpath:/default/delays_20141114.csv`
and
`classpath:/default/delays_lightpulser_20150217.csv`

contain delays, but they use a different sign. So these files **cannot** be interchanged. 
However both files are apparently used in xmls in the master. Might be dangerous.

Since the FACT hardware changes over time, I guess the 2014 file describe one epoch and the 2015 file another, but in that case, the numbers should *mean* the same, right?

-----

Bonus question: 

I tried to look into the files using `pandas.read_csv(path)` but it turned out I had to use:

    pd.read_csv("delays_20141114.csv", header=None, names=["delay"], skiprows=1)

and: 

    pd.read_csv("delays_lightpulser_20150217.csv", header=None, names=["delay"])

Wouldn't it be nice, if we formatted our csv files in a way, that is all nicely readable by fact-tools and `pandas.read_csv` and a human who happens to look at it in an editor?

dneise, 2016-07-28:
Ah and another question:

These delays might be measured in nanoseconds or slices. I don't know. I guess it would be good to write it into the file.

dneise, 2016-07-28:
Ah and another question:

These delays may have a global offset, which would probably not matter at all.
But it wouldn't hurt to define it, I guess.

The mean of the 2015 file is almost zero, but not the mean of the 2014 file. 
So maybe the mean of the delays except the delays of the broken pixel is zero? 
I don't know. 
I guess it should be written into the header of the file as well.

MaxNoe, 2016-07-28:
Seems like i fucked things up in abf94170a243521092cdfb5d97cb8b51e76d57b4

I switched the sign in that commit, and uploaded a file `classpath:/delays_20141114_subtract.csv` which should have gone into `classpath:/default/`, the file with the old sign is even there twice.

mackaiver, 2016-08-10:
Do we expect these delays to shift significantly over time? I don't quite understand why. I probably just forgot where these come from. 
In other words: can we just keep one?

dneise, 2016-08-10:
I think this is again a matter of so called epochs:
Hardware changes sometimes (spares for broken boards, ultimatively broken stuff, realigned mirrors, ...)
And sometimes just our understanding of hardware changes.

I think software should be kind of epoch-aware, i.e. *if* so called constants change several times in the future this shouldn't surprise devs and users. Also I think such changes should insert themselves quite naturally into this analysis software, and not make a total rewrite of several processors necessary.

TLDR:

It will change. Be prepared, i.e. provide a clear naming scheme for these resource files, which tells everybody (people and code) what the heck is going on with these files.

Maybe: `pixel_delays_valid_from_yyyymmdd_hhmmss.csv`

mackaiver, 2016-08-10:
Absolutely true. What should happen when someone uses an outdated file? Should there be a warning? Or maybe we shouldn't let the user choose at all and make it automatic?

dneise, 2016-08-10:
I'd say make it automatic, therefore filenames should contain human and machine readable dates ... I hope that is enough. So people browsing that folder have a chance to guess.

But also allow explicit specification of url key for explicitely testing different versions.

Big question: How does it work in detail?

open raw-data-file look at start date ... decide what ressources to use ... and go on?
What if the end date implies a different ressource should be used? Do we care?

mackaiver, 2016-08-10:
> Big question: How does it work in detail?

Implemtentation wise I guess we can use something similar to the current auxiliary services right? 
In any case I will prepare some more documentation somewhere.

dneise, 2016-08-10:
Is there some kind of beginners tutorial into the whole idea around aux services? 
I don't understand them at all ...

mackaiver, 2016-08-10:
> Is there some kind of beginners tutorial into the whole idea around aux services?
> I don't understand them at all ... 

No. Not yet. Thats what I meant. 

> In any case I will prepare some more documentation somewhere.

ftemme, 2016-08-10:
I already started something like the aux service. The CalibrationService (fact.calibrationservice).
Currently it contains a list of the bad pixels and a list of unusuable pixels (the broken bias patch) for different time periods.
The time periods are hardcoded in the ConstantCalibService (the current implemented CalibrationService). I had in mind that this CalibrationService stuff could be improved in the feature (maybe reading hardware epochs from a file or a database, or something like that).

MaxNoe, 2016-08-10:
The main difference is in the way those delays were calculated.

The 2014 files was calculated from Muon events, the 2015 lightpulser file, you might have guessed it, from lightpulsers.

dneise, 2016-08-10:
While this is true, it shouldn't matter. 
Either both of them can be **used** for the same thing or not, 
no matter how exactly the numbers were achieved. 

I think, users and fellow devs should be allowed to abstract away from such details as to how the numbers were created, and more concentrate on how to use them and when.

Of course, this doesn't mean the details on how these numbers were created should be impossible to find when needed, as it is the case in many places of another project I know. 
Say somebody uses a certain file that *should* work, but finds out it doesn't work. This person should in my opinion be enabled to find out how the numbers were created in the first place. 
I think in csv files a commented out header section is a great place to put links or further information on how in detail the resource file in question was created, or maybe just the email of the original author, if time is pressing.
(Is not needed in a git repo, since the original author can be found by the commit, but what if all I have is the jar-file?)

commit 143bc0ca (2017-11-20): Remove obsolete delay files, fixes #137
commit 57219e49 (2017-11-20): Merge pull request #262 from fact-project/delay_files

Remove obsolete delay files, fixes #137
closed 2017-11-20 by MaxNoe