fact-project/fact-tools #142: BufferUnderFlowException while reading fz file

MaxNoe, 2016-08-17:
Trying to read the file `20160817_017.fits.fz` I get

```
Aug 17 2016 20:15:03 [ERROR] stream.runtime.AbstractProcess: Aborting process due to errors: null
java.nio.BufferUnderflowException
        at java.nio.HeapByteBuffer.get(HeapByteBuffer.java:151)
        at java.nio.ByteBuffer.get(ByteBuffer.java:715)
        at fact.io.zfits.BlockHeader.unHuffman(BlockHeader.java:120)
        at fact.io.zfits.BlockHeader.decode(BlockHeader.java:162)
        at fact.io.zfits.ZFitsTableReader.readNextRow(ZFitsTableReader.java:74)
        at fact.io.zfits.ZFitsStream.readNext(ZFitsStream.java:193)
        at stream.io.AbstractStream.read(AbstractStream.java:135)
        at stream.runtime.AbstractProcess.execute(AbstractProcess.java:193)
        at stream.runtime.ProcessThread.run(ProcessThread.java:123)
```

MaxNoe, 2016-08-17:
Reading the much smaller (5 vs 50k Events) File `20160817_016.fits.fz` I get 

```
Aug 17 2016 20:20:16 [ERROR] stream.runtime.AbstractProcess: Aborting process due to errors: 5
java.lang.ArrayIndexOutOfBoundsException: 5
	at fact.io.zfits.ZFitsTableReader.readNextRow(ZFitsTableReader.java:89)
	at fact.io.zfits.ZFitsStream.readNext(ZFitsStream.java:193)
	at stream.io.AbstractStream.read(AbstractStream.java:135)
	at stream.runtime.AbstractProcess.execute(AbstractProcess.java:178)
	at stream.runtime.ProcessThread.run(ProcessThread.java:123)
```

mackaiver, 2016-08-18:
I cannot find these at ISDC and I cannot access the FHGFS for some reason. Can you attach the small file here?

MaxNoe, 2016-08-18:
The data is not yet transferred yet. You can find both files in `vollmond:/home/mnoethe/timelapse_calibration`

relleums, 2017-01-25:
I have the same issue for 2015/03/xx:

```java.nio.BufferUnderflowException
        at java.nio.HeapByteBuffer.get(HeapByteBuffer.java:151)
        at java.nio.ByteBuffer.get(ByteBuffer.java:715)
        at fact.io.zfits.BlockHeader.unHuffman(BlockHeader.java:120)
        at fact.io.zfits.BlockHeader.decode(BlockHeader.java:162)
        at fact.io.zfits.ZFitsTableReader.readNextRow(ZFitsTableReader.java:74)
        at fact.io.zfits.ZFitsStream.readNext(ZFitsStream.java:197)
        at stream.io.AbstractStream.read(AbstractStream.java:135)
        at stream.runtime.AbstractProcess.execute(AbstractProcess.java:193)
        at stream.runtime.ProcessThread.run(ProcessThread.java:123)
```

Here is a table of runs that mostly failed with this error.

| status | Night_Run |
| ----- | ------------ |
|       | 20150316_077 |
| error | 20150316_080 |
| error | 20150316_081 |
| error | 20150316_082 |
| error | 20150316_083 |
| error | 20150316_086 |
| error | 20150316_087 |
| error | 20150316_088 |
| error | 20150316_089 |
| error | 20150316_092 |
|       | 20150316_093 |
| error | 20150316_094 |
| error | 20150316_095 |
| error | 20150316_096 |
| error | 20150316_106 |
| error | 20150316_107 |
| error | 20150316_108 |
| error | 20150316_109 |
| error | 20150316_112 |
| error | 20150316_113 |
| error | 20150316_114 |
| error | 20150316_115 |
| error | 20150316_118 |
| error | 20150316_119 |
| error | 20150316_120 |
| error | 20150316_121 |
| error | 20150316_131 |
| error | 20150316_132 |
| error | 20150316_133 |
| error | 20150316_134 |
| error | 20150316_137 |
| error | 20150316_138 |
| error | 20150316_139 |
| error | 20150316_140 |
| error | 20150316_143 |
| error | 20150316_144 |
| error | 20150316_145 |
| error | 20150316_146 |
| error | 20150316_156 |
| error | 20150316_157 |
|       | 20150316_158 |
| error | 20150316_159 |
| error | 20150316_160 |
| error | 20150316_163 |
| error | 20150316_164 |
| error | 20150316_165 |
| error | 20150316_166 |
| error | 20150316_169 |
| error | 20150316_170 |
|       | 20150316_173 |
| error | 20150316_174 |
| error | 20150316_175 |
| error | 20150316_176 |
| error | 20150316_186 |
| error | 20150316_187 |
| error | 20150316_188 |
| error | 20150316_189 |
| error | 20150316_192 |
| error | 20150316_193 |
| error | 20150316_194 |
| error | 20150316_195 |
| error | 20150316_198 |
| error | 20150316_199 |
| error | 20150316_200 |
|       | 20150316_201 |
| error | 20150316_211 |
| error | 20150316_212 |
| error | 20150316_213 |
|       | 20150316_214 |
|       | 20150316_217 |
|       | 20150316_218 |
| error | 20150316_219 |
| error | 20150316_220 |
|       | 20150316_221 |
| error | 20150316_224 |
|       | 20150316_225 |
|       | 20150316_226 |

MaxNoe, 2017-01-26:
Are you experiencing the same with the new `fact.io.hdureader.FITSStream`?

relleums, 2017-01-26:
I have not tested this yet, but I am about to submit new jobs soon for this purpose.

closed 2017-06-13 by MaxNoe