fact-project/fact-tools #150: skip 27sec events

dneise, 2016-10-12:
We (Zurich) think the 27sec events should be skipped entirely during analysis.

An event is a 27sec event if many pixels are saturated for > 100 slices.
(we assume, if one pixel saturates for >100 slices ... then all do ... so maybe checking one pixel is already enought... but more are better.

We think flagging these files as "Is27SecondEvent" would be good.

We further think in Jens `CorrectSaturation` processor is well suited to find these.

jebuss, 2016-10-13:
In principle i think that this is reasonable. However, we I assume that these events won't survive the background suppression. Therefore I guess it would not make a difference weather we keep them or not. Or do you see some issue besides the disposal of runtime? I only wonder if the early suppression of these events might have some systematics with regards to none 27s-Events

dneise, 2016-10-13:
Ah, we talked a bit further and now agreed that simply flagging them as 27sec event is enough.

If it gets killed in a cut, fine.If not, it can be found.

commit 1de9202a (2018-01-04): Skip 27s events, fixes #150
commit f9c0687d (2018-01-04): Merge pull request #305 from fact-project/flag_27s

Skip 27s events, fixes #150
closed 2018-01-04 by MaxNoe