fact-project/fact-tools #156: NoSuchElementException ... half price today

dneise, 2016-10-18:
There are also some NoSuchElementExceptions available, in case you needs some... help yourself.

These files offer them...
```
['20140101_152',
                                                   '20140106_253',
                                                   '20140113_258',
                                                   '20140121_102',
                                                   '20140203_059',
                                                   '20140211_195',
                                                   '20140211_198',
                                                   '20140211_199',
                                                   '20140318_046',
                                                   '20140318_047',
                                                   '20140319_032',
                                                   '20140319_037',
                                                   '20140319_038',
                                                   '20140319_050',
                                                   '20140319_053',
                                                   '20140320_032',
                                                   '20140320_033',
                                                   '20140320_043',
                                                   '20140320_044',
                                                   '20140320_045',
                                                   '20140320_046',
                                                   '20140320_050',
                                                   '20140320_051',
                                                   '20140320_052',
                                                   '20140320_053',
                                                   '20140320_125',
                                                   '20140321_066',
                                                   '20140321_078',
                                                   '20140321_081',
                                                   '20140322_033',
                                                   '20140322_034',
                                                   '20140322_046',
                                                   '20140322_072',
                                                   '20140322_075',
                                                   '20140322_077',
                                                   '20140322_083',
                                                   '20140322_084',
                                                   '20140322_086',
                                                   '20140323_023',
                                                   '20140323_024',
                                                   '20140323_025',
                                                   '20140323_032',
                                                   '20140323_075',
                                                   '20140323_152',
                                                   '20140323_163',
                                                   '20140323_168',
                                                   '20140323_170',
                                                   '20140323_171',
                                                   '20140324_064',
                                                   '20140324_065',
                                                   '20140324_173',
                                                   '20140324_174',
                                                   '20140326_150',
                                                   '20140326_171',
                                                   '20140327_040',
                                                   '20140327_043',
                                                   '20140327_046',
                                                   '20140327_048',
                                                   '20140327_049',
                                                   '20140327_050',
                                                   '20140327_055',
                                                   '20140327_067',
                                                   '20140404_031',
                                                   '20140404_033',
                                                   '20140406_155',
                                                   '20140407_132',
                                                   '20140407_133',
                                                   '20140407_134',
                                                   '20140407_146',
                                                   '20140407_147',
                                                   '20140407_151',
                                                   '20140407_153',
                                                   '20140407_154',
                                                   '20140407_159',
                                                   '20140408_131',
                                                   '20140408_132',
                                                   '20140408_143',
                                                   '20140408_144',
                                                   '20140408_150',
                                                   '20140409_097',
                                                   '20140409_098',
                                                   '20140409_099',
                                                   '20140409_103',
                                                   '20140409_105',
                                                   '20140409_108',
                                                   '20140409_109',
                                                   '20140409_112',
                                                   '20140409_113',
                                                   '20140409_114',
                                                   '20140417_065',
                                                   '20140417_078'],
```

from stdout:
```
Oct 18 2016 05:16:36 [INFO ] fact.io.zfits.ZFitsStream: Reading file: /fact/raw/2014/04/17/20140417_078.fits.fz
Oct 18 2016 05:16:36 [INFO ] fact.io.zfits.ZFitsStream: File is ZFITS compresssed.
Oct 18 2016 05:16:36 [INFO ] fact.io.zfits.ZFitsStream: File contains ZDrsCellOffsets.
Oct 18 2016 05:16:37 [INFO ] stream.runtime.ProcessContainer: Initializing life-cycle objects...
Oct 18 2016 05:16:37 [INFO ] stream.runtime.ProcessContainer: Creating 1 active processes...
Oct 18 2016 05:16:38 [WARN ] fact.auxservice.AuxFileService$1: The file /fact/aux/2014/04/17/20140417.fits is not a recognized aux service. Could not parse file name into a recognized service.
Oct 18 2016 05:16:38 [INFO ] fact.io.FitsStream: End of file reached. /fact/aux/2014/04/17/20140417.DRIVE_CONTROL_SOURCE_POSITION.fits
Oct 18 2016 05:16:39 [INFO ] fact.io.FitsStream: End of file reached. /fact/aux/2014/04/17/20140417.DRIVE_CONTROL_TRACKING_POSITION.fits
Oct 18 2016 05:16:39 [INFO ] fact.io.zfits.TileHeader: Id of the TileHeader is wrong, got: '^Q?^N?'
Oct 18 2016 05:16:40 [ERROR] stream.runtime.AbstractProcess: Aborting process due to errors: The given ordering: '^@' byte: '0' is not supported
Oct 18 2016 05:16:40 [INFO ] stream.runtime.ProcessContainer$2: Process DefaultProcess['1'] finished!
Oct 18 2016 05:16:40 [INFO ] stream.run: Container finished.
```

stderr:
```
java.util.NoSuchElementException: The given ordering: '^@' byte: '0' is not supported
        at fact.io.zfits.BlockHeader$Ordering.getOrderingFormCharacter(BlockHeader.java:49)
        at fact.io.zfits.BlockHeader.<init>(BlockHeader.java:77)
        at fact.io.zfits.BlockHeader.<init>(BlockHeader.java:62)
        at fact.io.zfits.ZFitsTableReader.readNextRow(ZFitsTableReader.java:73)
        at fact.io.zfits.ZFitsStream.readNext(ZFitsStream.java:193)
        at stream.io.AbstractStream.read(AbstractStream.java:135)
        at stream.runtime.AbstractProcess.execute(AbstractProcess.java:193)
        at stream.runtime.ProcessThread.run(ProcessThread.java:123)
```

----

Looks to me, like the input file is broken. If it was broken, the callisto files for these files cannot exist, or must be very small. At least for one of the files, the callisto file exists (20140417_078). So maybe this is an issue with the zfits reader ... no idea.

mackaiver, 2016-10-20:
Not sure. I've tried figuring out whats going wrong. The TileHeader and BlockHeader bytes I looked at look weird. From what I can gather in Figures 5 and 6 in https://arxiv.org/pdf/1506.06045v1.pdf. The tile header should be starting with the ASCII text 'TITLE'. For some reasons there is just mumbo jumbo. Not sure yet where the error is. I'll keep looking.

dneise, 2016-10-20:
Maybe we can track the problem down, by comparing your reader and this: https://github.com/fact-project/zfits/

This python zfits reader is clearly not good, I just wanted to use it, and immediately found a stupid typo-kind-of bug. However I tried this and it worked:

```{python}
import zfits
f = zfits.ZFits("/fact/raw/2014/01/01/20140101_152.fits.fz")
N = f[2].get_nrows() # nuber of events
first_event = f.get_raw_data(0) 
last_event = f.get_raw_data(N-1) # events start counting at zero.
```

dneise, 2016-10-20:
However the zfits python reader uses esheldons fitsio, which is fortran-style-C .. so that part we cannot really review, I think.

mackaiver, 2016-10-20:
Okay from what I can gather the 'Catalog' portion of the files is broken (At least for the two files in the list that I've tried). 
Heres the catalog (offsett_in_bytes:size_of_following_data_chunk_in_bytes) for five rows of a healthy catalog:

```

//    Tiles: 3474, cols: 10
//            16:16,32:16,48:14,62:16,78:20,98:172,270:2892,3162:332,3494:267452,0:0,
//            270962:16,270978:16,270994:14,271008:16,271024:20,271044:172,271216:2892,274108:332,274440:266230,0:0,
//            540686:16,540702:16,540718:14,540732:16,540748:20,540768:172,540940:2892,543832:332,544164:268471,0:0,
//            812651:16,812667:16,812683:14,812697:16,812713:20,812733:172,812905:2892,815797:332,816129:265169,0:0,
//            1081314:16,1081330:16,1081346:14,1081360:16,1081376:20,1081396:172,1081568:2892,1084460:332,1084792:306704,0:0,

``` 
Summing up all size_of_following_data_chunk_in_bytes in a row and adding 32 bytes of padding you get the offset of the first chunk in the follwing row.

And heres the same for a 'unhealthy' catalog in 20140211_195

```
//
//    Okt 20 2016 19:23:23 [INFO ] fact.io.zfits.ZFitsTableReader: Tile Header: TILE, ID: 'TILE', numRows: 1, size: 1381489
//    Tiles: 26582, cols: 10
//            16:16,32:16,48:14,62:16,78:20,98:172,270:2892,3162:332,3494:267141,0:0,
//            792719:16,792735:16,792751:14,792765:16,792781:20,792801:172,792973:2892,795865:332,796197:259408,0:0,
//            1567409:16,1567425:16,1567441:14,1567455:16,1567471:20,1567491:172,1567663:2892,1570555:332,1570887:258586,0:0,
//            2349321:16,2349337:16,2349353:14,2349367:16,2349383:20,2349403:172,2349575:2892,2352467:332,2352799:253690,0:0,
//            3121114:16,3121130:16,3121146:14,3121160:16,3121176:20,3121196:172,3121368:2892,3124260:332,3124592:253140,0:0,

```

This is not the case. Look at the difference between line 1 and 2.

dneise, 2016-10-24:
ah .. have a look at the FITS keyword "ZSHRINK". This tells you how meesed up the whole thing is. For many files the value of ZSHRINK is 1, which means "moderately messed up" or "can be read by normal people". 
For the file you looked at (20140211_195) ZSHRINK is 3, which translates to "totally messed up" or only theoreticians of 5th level can read this file ...

dneise, 2016-10-24:
I **said** the python reader in fact-project/zfits could read the file, but that's only partially true (or total nonsense, whatever you prefer).

The header of 20140211_195 says, e.g.:
ZNAXIS2 = 79745 / Number of uncompressed rows 

while my zftis readers says: 
```
  file: /fact/raw/2014/02/11/20140211_195.fits.fz
  extension: 2
  type: BINARY_TBL
  extname: Events
  rows: 26582
```

So the number of rows, my reader sees is only about one third of the number of rows, the file should have.
And ZSHRINK is 3... coincidence? no way! In the paper it says:

> ZSHRINK.
In our Tile-compressed FITS streamer the main data table is used to store pointers to the heap-area, usually one for each compressed row.  Since this table is stored before the heap-area, it implies that the number of rows to be written to the  file is known in advance.  This is not always possible because experiments often prefer to group data per interval of time rather than per unit of raw data. As  a  remedy,  an  estimate  of  the  expected  number  of rows/events  is  calculated  when  the file  is  opened,  and  a corresponding number of bytes reserved.  If the number of events grows larger than the number of pointers which can be written to the table, only every N-th pointer is written to the table.  The integer N is then stored in the keyword ZSHRINK.

> To still be able to read the compressed data entirely, special markers are put in the heap area that allow us to not only read the compressed data without access to the main data table, but also to reconstruct this table entirely from the heap area only

dneise, 2016-10-24:
So this ZSHRINK != 1 case should be taken into account.

MaxNoe, 2016-10-24:
Ah, this makes a lot of sense. Not implementation wise, but it explains why we only now see this effect. This only affects files which are significantly larger than the normal files, right? Normally one of the first data selection steps included throwing away runs with unusual high trigger frequencies.

dneise, 2016-10-24:
Jepp, what makes no sense to me is: Why the facttools implementation, unlike the python implementation does not simply see too few events, but claims the file to be broken.

mackaiver, 2016-10-25:
So right now the Java ZfitsReader does something like this:

``` 
public readNextRow(){
     for (int col : numberOfColumnsInTable){
         int bytesToReadFromFile = Catalog.numberOfBytesForColumn(col, currentRow);
         byte[] columnData = readBytes(bytesToReadFromFile);
         BlockHeader header = parseBlockHeader(columnData);
     }
}
```

The Catalog is build once when the File is opened. It does not take the ZSHRINK keyword into account. We would need to change that.

dneise, 2016-10-25:
Is there also somewhere something like this

    current_file.seek( Catalog.startAddressOfColumn(col, currentRow);

If not, this could explain the error.

dneise, 2016-10-25:
I have some comments:

----
 
[In TileHeader.java line 17](https://github.com/fact-project/fact-tools/blob/master/src/main/java/fact/io/zfits/TileHeader.java#L17) it says `numRows` was  `number of rows pooled in this tile`, according to [Fig 6 in the only documentation we have](https://arxiv.org/pdf/1506.06045v1.pdf)

This **sounds**  like `numRows` might be the same thing as `ZSHRINK`, which is the number of events which can be reached by jumping their first event, but I think it is not.

----

Figure 4 in the only documentation we have is wrong. It looks as if there were addresses/pointers inside the "catalog" pointing to the "TILE" header. This is to my understanding not the case. All pointers, I've ever seen were pointing to "block headers", c.f. Figure 5.

----

The figure subscript of Fig 5 is wrong: It says:
> *processings*(*numprocs* bytes) is the identifiers of the applied algorithms.

Correct would be:

*processings*(*numprocs* **shorts**) is the identifiers of the applied algorithms.

This is correctly done in [BlockHeader.java line 82](https://github.com/fact-project/fact-tools/blob/master/src/main/java/fact/io/zfits/BlockHeader.java#L82) but has never founds its way into our documentation. Cost me a couple of days to find it, when I worked on the python implementation.

MaxNoe, 2017-06-13:
Solved by using the new `fact.io.hdureader`

closed 2017-06-13 by MaxNoe