fact-project/fact-tools #158: SinglePe extractor template maybe outdated

jebuss, 2016-10-28:
In the current version of fact.features.singlePulse.timeLineExtraction.TemplatePulse we use a template according to an old ceres.rc (The one from the Mars Trunk in svn). It looks like this:

`PulseShape.Function.Name:  (1.626*(1-exp(-0.3803*0.5*x))*exp(-0.0649*0.5*x))`

In ceres12.rc the following template is used:

`PulseShape.Function.Name:  (1.239*(1-1/(1+exp((0.5*x-2.851)/1.063)))*exp(-(0.5*x-2.851)/19.173))`

We should check the impact of these different functions to the extractors performance

dneise, 2016-10-31:
I wanted to understand the difference, so I plotted them for me ... maybe this is interesting for other people as well:
[image]

The blue is the upper fomula, the green is the lower formula.

relleums, 2016-11-15:
I opt for exploring different pulse template shapes when we managed to close the loop with the simulation truth on the pulses from CERES/MARS. But it is good that we noticed this mismatch of the different pulse templates.