fact-project/fact-tools #159: SinglePe extractor, AC coupling bug

relleums, 2016-10-28:
The current single pe extractor AC coupling (which performs a subtraction of the mean of the overall amplitude time line) has no effect yet. Jens and I made the subtraction work, but this created artefacts in about 1% of the time lines. This was not yet seen in the single pulse extractor sandbox. The probability of these artefacts might be reduced with a better pulse template, but we are afraid in can not be ruled out. Maybe the algorithm needs modifications here. The good thing however is, that there seems to be better reconstruction of single photon pulses.

jebuss, 2016-10-28:
[image]

This ist THE artefact Max was talking about

dneise, 2016-10-31:
There is a [stop criteria](https://github.com/fact-project/fact-tools/blob/master/src/main/java/fact/features/singlePulse/timeLineExtraction/SinglePulseExtractor.java#L96) in the algorithm. I think one can add a crititerium, which checks if `maxResponse` is decreasing or not.

As far as I understood, `maxResponse` should always decrease and never increase. In case this artefact is produced, it looks like it increases.

One might think this is a too crude way to get this artefact under control. In that case maybe one can use at least look at the change of `maxResponse` from one iteration to the next in order to find the actual reason for the artifact and then use a more sophisticated method to avoid the generation of such artefacts. 

Do you know if the single pulses, which are found **before** the artifact is starting to be generated are kinda correct and "enough", if you know what I mean?

relleums, 2016-10-31:
You are right Dominik, I explained that the maxResponse correlates to the number of pulses stacked on top of each other. However, I learned myself now that this is not necessarily true because of the mean subtraction after each iteration. Anyhow, for now we will focus on getting the MC truth available to the evaluation. Otherwise it is very tedious to improve the Single Pulse Extractor in its fact-tools environment.
In the very early steps in the python sandbox of the single Pulse Extractor, I used a different mechanism to compensate for the baseline floating which was a little bit more computational expensive, but did not show advantages in the sandbox over the subtraction of the mean. I will give this first method a try again.

closed 2016-11-25 by jebuss