fact-project/fact-tools #179: AuxPoints contain 0 values although no 0s are in the file

MaxNoe, 2016-12-16:
This leads to strange behaviour, since the time is then 1970-1-1

MaxNoe, 2016-12-16:
This seems to happen for the last 4 rows of every aux file. 

The FITSStream puts zeros for all keys.

dneise, 2016-12-16:
I've seen 0s in the Time column of aux files yesterday while looking at rates with astropy.Table.read. So at least in some files there are in fact zeros as Times... 

Apparebtly a bug somewhere in FACT++ nobody bothered to report or fix so far, including me of course.

You sure, the file u talking about might not be one of those suffering from said bug?

dneise, 2016-12-16:
Oh okay

MaxNoe, 2016-12-16:
I checked the files with astropy.table.read and there are no rows with all zeros in the test file.

mackaiver, 2016-12-19:
Can you give an example file?

MaxNoe, 2016-12-19:
Our test data aux files.

MaxNoe, 2017-01-09:
This seems to be a problem of the `fact.io.FitsStream`. I ported the AuxService to our new HDUReader here: #181 and the problem seems to be gone.

mackaiver, 2017-01-26:
I guess this can be closed now.

MaxNoe, 2017-01-26:
The corresponding PR is not yet merged. It still has conflicts that need to be resolved

MaxNoe, 2017-11-07:
Should be solved by now

closed 2017-11-07 by MaxNoe