fact-project/fact-tools #182: single pulse extractor output is not yet size efficient, baselines too fine

relleums, 2017-01-16:
File size comparison for FACT run 20151001_011 with fact-tools 0.17.2 pass2    

| file type | size [MB] | reduction [rel.] |
| ------------- | ------------- |-----------|
| zfits  | 2935  | 1.0 |
| fact-tools jsonl.gz  | 305 | 9.6 |
| optimized jsonl.gz  | 154 | 19.1 |
| optimized and truncated time lines jsonl.gz  | 94 | 31.2 |
| optimized, truncated time lines, and no baselines jsonl.gz  | 69 | 42.5 |

The baselines (useful for time series reconstruction) are a mayor contribution to the output size as they are stored as floats:
```"PhotonArrivalsBaseLine":[-0.9346934899690481,-0.914864300604518, ...```

This may be optimized by storing the baseline in e.g. centi 1pe amplitudes which is still fine enough:
```"PhotonArrivalsBaseLine100":[-93,-91, ...```

dneise, 2017-01-16:
started working on it, c.f https://github.com/fact-project/fact-tools/commit/da29965ddc243c01f6140dc4aa5754ca6e0ea935

MaxNoe, 2017-01-16:
The facttools JSONWriter has the option `doubleSignDigits` which sets the number of decimal places for doubles.

dneise, 2017-01-16:
yes, but this is globally affecting the precision, right? The proposal is to only change this for the BaseLines

MaxNoe, 2017-01-16:
Yes, i just wanted to mention it in case you were not aware of the option. (And I don't think we have anything that would require the default of 8 digits)

dneise, 2017-01-17:
closed by #186

closed 2017-01-17 by dneise