fact-project/fact-tools #190: Export list of saturated pixels in the single pulse extraction passes

relleums, 2017-01-17:
(Partly on behalf of Adrian)

We propose a variable length list in the output which stores the ids of saturated pixels.
Such a list would be most often empty (size efficient).

```"SaturatedPixels:[81,123,1439]" ```

For the single pulse extractor passes we want to not run the pulse extraction on exceptional bright events e.g. 27s events or LIDAR. Adrian proposes to not extract single pulses when a certain fraction of the pixels is saturated (maybe 10%?).

Adrian highly recommends to store such a list in the output of the single pulse extraction passes. Although these are again detector specific values, the list would at least be empty for most events. This way one can decide if one wants to process events with saturated pixels. Further we might prevent running into the wall time. Bright events were no extraction was performed simply do not have a photon_stream in them.

Could somebody point us to a sample run with  exceptional bright events?

relleums, 2017-01-17:
This turns out to be a duplicate or extension to issue #150

dneise, 2017-01-17:
This guy already outputs such a list...
but it is not used anymore at all... does this mean it does not work?

https://github.com/fact-project/fact-tools/blob/master/src/main/java/fact/extraction/HandleSaturation.java

see also #192

dneise, 2017-01-17:
Ah no ... this `TimeOverThreshold` guy seems to output such a list by the "timeOverThresholdSet" key.

dneise, 2017-01-17:
In the XML I would propose to do it like this:

    <fact.pixelsets.Length
      pixelSetKey="timeOverThresholdSetOverLay"
      outputKey="NumberOfSaturatedPixels"
    />
    <stream.flow.If 
        condition="%{data.NumberOfSaturatedPixels} &lt; 100"
    >
        <fact.extraction.SinglePulseExtraction
            dataKey="DataCalibrated"
            outputKey="PhotonArrivals"
        />
    </stream.flow.If>

MaxNoe, 2018-03-02:
This was solved in the commits linked above

closed 2018-03-02 by MaxNoe