fact-project/fact-tools #193: treat also broken bias patch 272

dneise, 2017-01-17:
Here: https://github.com/fact-project/fact-tools/blob/master/src/main/resources/default/settings.properties#L13

we state all broken pixel. Since 13.03.2016 or so also the bias patch 272 is broken.
The CHIDs of these pixels are: 1296, 1297, 1298, 1299.

Here I wrote something about it including all one million different pixel ids I could find.
https://www.fact-project.org/logbook/showthread.php?tid=4830&pid=23739#pid23739

dneise, 2017-01-17:
How to include also these broken pixels? 
I assume we need some kind of service, the processors can ask for stuff, and depending on time (i.e. night and run) this service would give them a different set of broken pixels.... 

Does something like that maybe already exist?

MaxNoe, 2017-01-17:
I think Fabian started some work in a Service which should deliver time based broken stuff information.

https://github.com/fact-project/fact-tools/blob/master/src/main/java/fact/calibrationservice/ConstantCalibService.java

There you could add a new HardwareConfiguration with additional broken pixels. These are then treated by the cleaning and extraction routines i think.

commit 853146ad (2018-02-16): Merge pull request #321 from fact-project/bias_patch_272

Refactor ConstantCalibService and HardwareConfiguration, fixes #193
commit 8b5a1c31 (2018-02-16): Refactor ConstantCalibService and HardwareConfiguration, fixes #193
closed 2018-02-16 by MaxNoe