fact-project/fact-tools #197: use "integralGains" for single pulse extraction?

dneise, 2017-01-19:
When extracting the `photoncharge` we use these individual pixel gain number:
https://github.com/fact-project/fact-tools/blob/master/src/main/resources/default/gain_sorted_20131127.csv
(btw. this is no CSV at all)
They look like this:
[image]

I was wondering if we should also use these numbers in the single pulse extractor, here:
https://github.com/fact-project/fact-tools/blob/master/src/main/java/fact/extraction/SinglePulseExtraction.java#L117

At the moment the constant `factSinglePeAmplitudeInMv` which is used here, is 10.
but we can also convert these integralGains from the file into this number.
The distribution for all pixels looks like this then:
[image]

jebuss, 2017-01-20:
Good idea, makes sense to me. Let's test it. However the amplitude might only be a 10% effect at max.

dneise, 2017-01-20:
Is it a good idea then? Or a waste of time? We have very very limit amout of time only. 
The single photon extraction pass should have been started since long. 

Question is not whether its nice or not ... surely one can do it and test it ... but is it necessary? 
Will people say: "This extraction is invalid", if we do not use these individual gains?

jebuss, 2017-01-20:
I don't know if it is necessary or not. But I think it would not make the result worse, only better! That's my guess. So put it in, if you agree. If we don't use it, we might misestimate Cherenkov pulses by up to 10%, since we apply a 10mv Amplitude to a 9mv pulse, which is also my guess.

dneise, 2017-01-20:
> since we apply a 10mv Amplitude to a 9mv pulse, which is also my guess.

Have a look at the lower image: 
https://cloud.githubusercontent.com/assets/8200858/22109410/e836dcae-de57-11e6-8661-4cd0b738c706.png

It shows the factors, which would be applied instead of the constant factor 10. ...

jebuss, 2017-01-20:
Yes that's what I meant.

jebuss, 2017-01-20:
I was talking about the worst case which is the outlier at 9mv

jebuss, 2017-01-20:
To shorten the discussion. I think it's a good idea. I think it's worth it. I thinking would improve the result. I think we should go for it.

dneise, 2017-01-20:
Adrian agreed with you

relleums, 2017-01-21:
I introduced a single pulse gain calibration service to avoid code duplication. However, I needed to modify ALL xmls.

MaxNoe, 2018-02-16:
Related to #267

MaxNoe, 2018-03-15:
The GainService is now in the master