fact-project/fact-tools #198: Create PhotonStream package

jebuss, 2017-01-19:
create a `fact.photonstream` package and move all processors that are related to the photon stream to this package, e.g.:

* `fact.extraction.SinglePulseExtraction`
* `fact.utils.PhotonStream2ArrivalTime`
* `fact.utils.PhotonStream2NumberOfPhotons``

relleums, 2017-01-19:
It is not only extraction any more. We also want to collect tools here for the reconstruction of photon charge and arrival time from the photon-stream.

what shall it be?
```
fact.PhotonStream.
fact.photonstream.
fact.photon_stream.
fact.photon-stream.
```

MaxNoe, 2017-01-19:
You missed lower-camel case: photonStream. I would go for upper camel case: `fact.PhotonStream`

dneise, 2017-01-20:
all other packages are *all lowercase*

jebuss, 2017-01-20:
Then we should stick to that

dneise, 2017-01-20:
so `photonstream` it is? no dash no underscore?

MaxNoe, 2017-01-20:
agreed.

PR #202 (2017-01-20): Create photonstream package
fixes #198
closed 2017-01-20 by MaxNoe