fact-project/fact-tools #209: Stopping to call stuff data and stopping to be cute

relleums, 2017-01-22:
I propose to rename all uses of the term ```data``` to ```observations``` when we refer to data which was taken by the actual FACT telescope and rename all uses of the term ```MC``` or ```Monte Carlo``` to ```simulations``` when we refer to data which was the outcome of a simulation of the FACT telescope. Since the term ```data``` has also a special meaning in the ```streams``` library (and a __completely arbitrary__ meaning in general) we should not use it for our belongs. 

I believe that this will help future students to get involved into the project. There are more variable names out there. Maybe they are not well explored yet but I am confident that we can find them and make use of these. Your variables have a personality, they have a story behind them. They have come around many lines or even files. Some of them even live on in the non-volatile nirvana of the ISDC. They have seen things, man. All this live and joy they lose when you call them ```data```. Your variables have deserved better than ```data```. 

```fact-tools/src/main/resources/default/observations/``` and ```fact-tools/src/main/resources/default/simulations/```

dneise, 2017-01-23:
11 and 12 letters ... they are even nearly of equal length ... 
I like that very much :-D

dneise, 2017-01-23:
Now seriously, this is a good proposal. But it is more than a fact-tools issue, it is a general issue.
And while this problem sounds huge and overwhelming sometimes, the way to the solution starts with the first step. So I vote this issue.

I would also like to ask all developers, who read this, to look at their own projects and this where they can refactor variable names. Also look at projects of others, one is sometimes blind to ones own mistakes, but can easily spot them in projects of others.

MaxNoe, 2017-01-23:
+1 for observations/simulations

MaxNoe, 2018-03-01:
This was done in #318

closed 2018-03-01 by MaxNoe