fact-project/fact-tools #212: viewer very resistent against nulls; kills; Ctrl-C; using it in general

dneise, 2017-01-25:
We can't kill the viewer anymore ... its here to stay ... 

doing this:
```
java -jar target/fact-tools-0.18.0.jar examples/viewer.xml
```

and hitting Ctrl-C to end it results in this:
```
Jan 25 2017 18:57:38 [INFO ] org.reflections.Reflections: Reflections took 82 ms to scan 1 urls, producing 56 keys and 288 values 
^CJan 25 2017 18:57:42 [INFO ] streams.runtime.Supervisor$1: The following consumers are attached to the root fact.io.hdureader.FITSStream@311d617d:  [DefaultProcess['2']@61064425]
```

relleums, 2017-01-25:
Exactly the same for me. Its here to stay.

MaxNoe, 2017-01-25:
This has to be #180

MaxNoe, 2017-01-26:
https://bitbucket.org/cbockermann/streams/issues/54/after-update-to-streams-10-fact-tools

commit 20c5a311 (2018-02-16): Make viewer killable again, fixes #212
commit 936862e8 (2018-02-16): Merge pull request #325 from fact-project/fix_viewer

Make viewer killable again, fixes #212
closed 2018-02-16 by MaxNoe