fact-project/fact-tools #216: Auxservice, NullPointerException

relleums, 2017-01-28:
In run 20140405_130
```
Exception in thread "Thread-1" java.lang.RuntimeException: Process stopped after exception: java.lang.NullPointerException
        at stream.runtime.ProcessThread.run(ProcessThread.java:170)
Caused by: java.lang.NullPointerException
        at fact.auxservice.AuxFileService.readDataFromFile(AuxFileService.java:92)
        at fact.auxservice.AuxFileService.getAuxiliaryData(AuxFileService.java:66)
        at fact.features.source.SourcePosition.process(SourcePosition.java:212)
        at stream.runtime.AbstractProcess.process(AbstractProcess.java:126)
        at stream.runtime.AbstractProcess.execute(AbstractProcess.java:184)
        at stream.runtime.ProcessThread.run(ProcessThread.java:129)
```

mackaiver, 2017-01-30:
For some reason I cannot login to the ISDC. Could you try #218 please?

relleums, 2017-01-30:
All errors when processing 1% (random) of all FACT runs.
I made a machine readable JSON collection of all the errors, the exact input files, and exact program calls:
[error_collection.json.txt](https://github.com/fact-project/fact-tools/files/738867/error_collection.json.txt)

PR #218 (2017-06-30): Aux fixes
Hello. 
Since the other two pull requests  #200 and #165 are somehow completely f#d up I just redid it here. Maybe this also fixes #216
closed 2017-06-30 by jebuss