fact-project/fact-tools #219: Runs passed without std. errors, but incomplete

relleums, 2017-01-31:
At least for the following runs:
20170124_093
20170123_185
20170105_047
I found that the fact-tools output has only 1/2 of the expected events. Usually this numbers are exact agreement with the run-info database. There was no stderror written by fact-tools for these runs.

dneise, 2017-02-01:
There is a line in BinTable.java:

    numberOfRowsInTable = header.getInt(“NAXIS2“).orElse(0);

I fear, NAXIS2 is not the number of events in a file if ZSHRINK is != 1.

So the reader is happyly reading the first half of the file, and then stopping, since it read as many events it thought it should read. No stderr, from its view all is fine.

Maybe one can check this hypothesis, and find a number in the header, which gives the true number of events.

Maybe ZNAXIS2 ?

dneise, 2017-02-01:
Unit tests on the short files we provide for testing, cannot find such a bug, since the files we provide are just a few bytes cut off a real file. So no unit test may read until the end of file...

Also artificially created zfits files containing ony a small number of events (assuming we had a zfits writer at hand) would not set ZSHRINK != 1, ... 

So testing this the way you did, ist maybe the only way...

relleums, 2017-02-01:
Indeed for 20170124_093 the value if ```ZNAXIS2``` is 48791 which is close to the expected number of events.

MaxNoe, 2017-02-17:
I checked #223 with file 20170123_047 and I find 42331  events. That's strange.

closed 2017-02-17 by MaxNoe

relleums, 2017-02-17:
I can not reproduce my 'expected event numbers'. I do not know what I did to get these. Since you found a match with the expected number of events in the database, I would say your numbers are correct and I did sth. wrong although I thought to be doing exactly what you did.

commit 36ecd210 (2017-02-17): Check ZNAXIS2 for number of events, fix #219
commit 94eafd1f (2017-02-17): Merge pull request #223 from fact-project/znaxis2_fix_219

Check ZNAXIS2 for number of events, fix #219