fact-project/fact-tools #238: fact.io.zfits.ZFitsStream.readNext io ERROR

relleums, 2017-07-06:
fact-tools: master, version 0.18.1, commit c49314cc68693e1e4758b1727714150c24ee9c9c
java: jdk1.8.0_111

Error reading this FACT observation run: ```20141110_133```
The error happened so far only once out of 1000 runs, so its not too crucial.

stdout:
```
Jul 06 2017 02:02:07 [INFO ] stream.util.PropertiesHandler: Reading properties from URL classpath:/default/settings.properties
Jul 06 2017 02:02:07 [INFO ] stream.util.PropertiesHandler: Reading properties from URL classpath:/default/settings.properties
Jul 06 2017 02:02:08 [INFO ] stream.runtime.ProcessContainer: Compute graph:

  {
   'sources': [
  	{ 'id': 'fact', 'class': 'fact.io.zfits.ZFitsStream' }
     ],
   'processes': [
  	{ 'id': '2', 'input': 'fact', 'class': 'stream.runtime.DefaultProcess' }
    ]
  }

Jul 06 2017 02:02:08 [INFO ] stream.run: Starting process-container...
Jul 06 2017 02:02:08 [INFO ] fact.io.zfits.ZFitsStream: Reading file: /data/fact_data/2014/11/10/20141110_133.fits.fz
Jul 06 2017 02:02:08 [INFO ] fact.io.zfits.ZFitsStream: File is ZFITS compresssed.
Jul 06 2017 02:02:08 [INFO ] fact.io.zfits.ZFitsStream: File contains ZDrsCellOffsets.
Jul 06 2017 02:02:10 [ERROR] stream.runtime.AbstractProcess: Error occurred in process '2': null
Jul 06 2017 02:02:10 [ERROR] stream.runtime.ProcessThread: java.nio.BufferUnderflowException
	at java.nio.HeapByteBuffer.get(HeapByteBuffer.java:151)
	at java.nio.ByteBuffer.get(ByteBuffer.java:715)
	at fact.io.zfits.BlockHeader.unHuffman(BlockHeader.java:120)
	at fact.io.zfits.BlockHeader.decode(BlockHeader.java:162)
	at fact.io.zfits.ZFitsTableReader.readNextRow(ZFitsTableReader.java:74)
	at fact.io.zfits.ZFitsStream.readNext(ZFitsStream.java:197)
	at stream.io.AbstractStream.read(AbstractStream.java:142)
	at stream.runtime.AbstractProcess.execute(AbstractProcess.java:194)
	at stream.runtime.ProcessThread.run(ProcessThread.java:129)

Jul 06 2017 02:02:10 [INFO ] stream.run: Container finished.
Jul 06 2017 02:02:10 [ERROR] stream.runtime.ProcessContainer$3: Process DefaultProcess['2']@7c30a502 signaled an error: null
```

stderror:
```
Exception in thread "Thread-2" java.lang.RuntimeException: Process stopped after exception: null
	at stream.runtime.ProcessThread.run(ProcessThread.java:170)
```

relleums, 2017-07-07:
Here are some more affected runs

NIGHT_RUN
```
grep -rnw 'BufferUnderflowException'
20141110_133 [ERROR] java.nio.BufferUnderflowException
20141113_210 [ERROR] java.nio.BufferUnderflowException
20141114_206 [ERROR] java.nio.BufferUnderflowException
20141114_208 [ERROR] java.nio.BufferUnderflowException
20141114_209 [ERROR] java.nio.BufferUnderflowException
20141124_180 [ERROR] java.nio.BufferUnderflowException
20141124_183 [ERROR] java.nio.BufferUnderflowException
20141124_197 [ERROR] java.nio.BufferUnderflowException
20141124_199 [ERROR] java.nio.BufferUnderflowException
20141213_177 [ERROR] java.nio.BufferUnderflowException
20141214_232 [ERROR] java.nio.BufferUnderflowException
20141214_250 [ERROR] java.nio.BufferUnderflowException
20141215_186 [ERROR] java.nio.BufferUnderflowException
```

mackaiver, 2017-07-07:
Hi did you try the `fact.io.hdureader.FITSStream`. ?

relleums, 2017-07-08:
Indeed, I am an idiot. Sorry for this. I did forget that the reader was replaced instead of being fixed, so I have to modify my steering cards accordingly. Btw, is there any feature of the 'old' reader that is not supported by the new one? Could we maybe just drop the old one?

closed 2017-07-08 by relleums