fact-project/fact-tools #239: :q

relleums, 2017-07-06:


relleums, 2017-07-07:
Here are some more affected runs:

NIGHT_RUN
```
grep -rnw 'NoSuchElementException'
20141124_184 [ERROR] java.util.NoSuchElementException: The given ordering: '￿' byte: '-1' is not supported
20141124_185 [ERROR] java.util.NoSuchElementException: The given ordering: '4' byte: '52' is not supported
20141124_186 [ERROR] java.util.NoSuchElementException: The given ordering: '' byte: '3' is not supported
20141124_208 [ERROR] java.util.NoSuchElementException: The given ordering: 'b' byte: '98' is not supported
20141124_210 [ERROR] java.util.NoSuchElementException: The given ordering: '' byte: '1' is not supported
20141211_161 [ERROR] java.util.NoSuchElementException: The given ordering: '￲' byte: '-14' is not supported
20141212_143 [ERROR] java.util.NoSuchElementException: The given ordering: '' byte: '3' is not supported
20141213_180 [ERROR] java.util.NoSuchElementException: The given ordering: '' byte: '1' is not supported
20141213_197 [ERROR] java.util.NoSuchElementException: The given ordering: 'v' byte: '118' is not supported
20141215_161 [ERROR] java.util.NoSuchElementException: The given ordering: 'ﾘ' byte: '-104' is not supported
20141215_190 [ERROR] java.util.NoSuchElementException: The given ordering: '￭' byte: '-19' is not supported
```

relleums, 2017-07-08:
Its my fault, I used the wrong FITS reader. Can we remove the old one?

closed 2017-07-08 by relleums

MaxNoe, 2017-07-08:
@relleums Yes, we should!