fact-project/fact-tools #240: fact.io.hdureader, IllegalArgumentException

relleums, 2017-07-08:
In run 20141124_209

stdout:
```
```

stderror:
```
java.lang.IllegalArgumentException: The value for THEAP should be equal to (NAXIS1*NAXIS2 + padding to 2880 blocks).
See section 7.3.2 of the FITS 3.0 standard
	at fact.io.hdureader.BinTable.binTableSanityCheck(BinTable.java:256)
	at fact.io.hdureader.BinTable.<init>(BinTable.java:129)
	at fact.io.hdureader.HDU.<init>(HDU.java:92)
	at fact.io.hdureader.FITS.<init>(FITS.java:91)
	at fact.io.hdureader.FITSStream.init(FITSStream.java:41)
	at stream.runtime.ProcessContainer.execute(ProcessContainer.java:619)
	at stream.runtime.ProcessContainer.run(ProcessContainer.java:572)
	at stream.run.main(run.java:171)
	at stream.run.main(run.java:144)
Exception in thread "main" java.lang.RuntimeException: java.lang.IllegalArgumentException: The value for THEAP should be equal to (NAXIS1*NAXIS2 + padding to 2880 blocks).
See section 7.3.2 of the FITS 3.0 standard
	at stream.runtime.ProcessContainer.run(ProcessContainer.java:575)
	at stream.run.main(run.java:171)
	at stream.run.main(run.java:144)
Caused by: java.lang.IllegalArgumentException: The value for THEAP should be equal to (NAXIS1*NAXIS2 + padding to 2880 blocks).
See section 7.3.2 of the FITS 3.0 standard
	at fact.io.hdureader.BinTable.binTableSanityCheck(BinTable.java:256)
	at fact.io.hdureader.BinTable.<init>(BinTable.java:129)
	at fact.io.hdureader.HDU.<init>(HDU.java:92)
	at fact.io.hdureader.FITS.<init>(FITS.java:91)
	at fact.io.hdureader.FITSStream.init(FITSStream.java:41)
	at stream.runtime.ProcessContainer.execute(ProcessContainer.java:619)
	at stream.runtime.ProcessContainer.run(ProcessContainer.java:572)
	... 2 more
```

MaxNoe, 2017-07-08:
I think this got fixed in #235, or did you use the newest version?

relleums, 2017-07-09:
I use:
fact-tools: master, version 0.18.1, commit c49314c
java: jdk1.8.0_111

MaxNoe, 2017-07-09:
Could you try again with the current master? https://github.com/fact-project/fact-tools/commits/master you are on the last commit before the mentioned PR

relleums, 2017-07-10:
Ah, ok. I missed the merge by one day or so. I will give the latest version a try soon.

relleums, 2017-07-11:
The latest master is now used to convert 2015/02 into photon-stream, so far about 1000 runs and not a single line in the std error. This is a clear improvement already. Thanks a lot for this. I will do further tests tomorrow where I will count the number of converted events in each run and compare to the (non ridiculous) part of the runinfo DB.

dneise, 2017-07-11:
there is one?

MaxNoe, 2017-11-07:
Seems to be solved

closed 2017-11-07 by MaxNoe