fact-project/fact-tools #244: DrsTimeCalibration throwing error in current master

MaxNoe, 2017-08-14:
```
$ java -jar target/fact-tools-0.18.1.jar examples/stdAnalysis/data/analysis.xml 
Aug 14 2017 19:07:17 [INFO ] stream.util.PropertiesHandler: Reading properties from URL classpath:/default/settings.properties
Aug 14 2017 19:07:17 [INFO ] stream.util.PropertiesHandler: Reading properties from URL classpath:/default/settings.properties
Aug 14 2017 19:07:18 [INFO ] stream.runtime.ProcessContainer: Compute graph:

  {
   'sources': [
  	{ 'id': 'fact', 'class': 'fact.io.hdureader.FITSStream' }
     ],
   'processes': [
  	{ 'id': '2', 'input': 'fact', 'class': 'stream.runtime.DefaultProcess' }
    ]
  }

Aug 14 2017 19:07:18 [INFO ] stream.run: Starting process-container...
Aug 14 2017 19:07:27 [ERROR] fact.filter.DrsTimeCalibration: Failed to load DRS data: mark/reset not supported
java.lang.RuntimeException: mark/reset not supported
	at fact.filter.DrsTimeCalibration.init(DrsTimeCalibration.java:43)
	at stream.runtime.AbstractProcess.init(AbstractProcess.java:144)
	at stream.runtime.ProcessThread.init(ProcessThread.java:115)
	at stream.runtime.ProcessContainer.execute(ProcessContainer.java:665)
	at stream.runtime.ProcessContainer.run(ProcessContainer.java:572)
	at stream.run.main(run.java:171)
	at stream.run.main(run.java:144)
Exception in thread "main" java.lang.RuntimeException: java.lang.RuntimeException: mark/reset not supported
	at stream.runtime.ProcessContainer.run(ProcessContainer.java:575)
	at stream.run.main(run.java:171)
	at stream.run.main(run.java:144)
Caused by: java.lang.RuntimeException: mark/reset not supported
	at fact.filter.DrsTimeCalibration.init(DrsTimeCalibration.java:43)
	at stream.runtime.AbstractProcess.init(AbstractProcess.java:144)
	at stream.runtime.ProcessThread.init(ProcessThread.java:115)
	at stream.runtime.ProcessContainer.execute(ProcessContainer.java:665)
	at stream.runtime.ProcessContainer.run(ProcessContainer.java:572)
	... 2 more
```

mackaiver, 2017-09-08:
Why the hell are there two processors with the same name?
```

fact.datacorrection.DrsTimeCalibration
fact.filter.DrsTimeCalibration

```

MaxNoe, 2017-09-08:
Once upon a time at a General Meeting in Zurich, @dneise had an idea on how to do the time calibration differntly and better. So @mackaiver  and @dneise started to implement the new version. It never got finished or tested. But the processor is still here in case someone some day might want to finish it.

commit 408c4efc (2017-11-15): Use BufferedInputStream, fixes #244
closed 2017-11-15 by MaxNoe