fact-project/fact-tools #255: Remove source package

MaxNoe, 2017-11-15:
We struggled for years to do source feature reconstruction right in java.

We failed basically, because precise coordinate transformations are hard.
We now do this using astropy in our high level analysis. Which is much more flexibel.

I would propose to remove the source package, so FACT-Tools is basically just used for calibration/extraction/source independent parameterisation.

What do you think?

dneise, 2017-11-15:
I totally agree

MaxNoe, 2017-11-17:
It seems we (or I) gave up too early. CoordinateTrafos are now precise enough for our use case (e.g. error is smaller than a pixel).

MaxNoe, 2018-01-15:
Kai and I decided against completely removing the code, but removing it from the standard analysis

closed 2018-01-15 by MaxNoe