fact-project/fact-tools #267: Consider changing gains in Extraction

jebuss, 2017-11-23:
Individual Pixels' gains do change over time. While some mail functions of pixels are considered in the Calibration Service, the variation of the gains is not.

In the worst case, the median camera gain varies by (30 mV slices) and the minimum gain even by (250 mV slices), in both cases this was determined on a window size of 30 slices and the generation of single pe spectra.

Here you see the time series of the gain measured in single pe spectras
[image]

This is the distribution of the median values of the time series above.
[image]

This is the distribution of the minima of the time series above.
[image]

I suggest to check if these variations have to be considered and if this is the case, a gain service for the extraction algorithm.

jebuss, 2017-11-23:
Right now I am checking how the single pe spectrum features of individual pixels change over time. Basically this would deliver the input for such a gain service.

dneise, 2017-11-23:
I have a question. What is the uncertainty on the gain, especially for such values like the  minimum gain. 
Is a gain of ~100...150mV slices still to be considered a relevant measurement?

jebuss, 2017-11-23:
Well thats a good question. 

Just for clarification, with minimum gain I mean the gain of the pixel with the lowest gain. 

So the uncertainty on each of theses values is the uncertainty of the gain determined from a single pixels spectrum made from 10000 triggers (aka events).

This uncertainty, I am sure how to determine that.

In an very old study I compared the gain determined from data and from MCs (see below). While the gain from data would have a systematic distribution over the pixels, for the MCs all pixel should have the same gain, since this is not varied and all pixels are simulated the same.

So judging from what you see below one could conclude that the std_dev from MCs is an rough estimate for the uncertainty of the method. Here you have a std_dev of ~3.5 mV slices calculates to an uncertainty of the mean value of  ~3.5 mV/sqrt(1439) ~ 0.9 mV slices.

[image]

jebuss, 2017-11-23:
> Is a gain of ~100...150mV slices still to be considered a relevant measurement?

Also here, I don't know for sure. At some point the maxima in the spe spectrum are too close together to distinguish between peaks, this could cause the fit of the spectrum model to fail. It's not clear to me at what minimum gain this happens. What we do have is the chi2 and ROOT's fit probability p-value. We could use those to reject gains were the fit did not work.

MaxNoe, 2017-12-01:
Should we add a fitsfile with rows for each time period? Or how would you propose to do this without adding the need for yet another file one has to find for each run?

PR #333 (2018-03-01): Gain service
Fixes #267 

* Introduces a new service which reads in a fits file with the gains of the single pe measurements
* Fits file in the resources with values from 20130825 -- 20170917
* If `UnixTimeUTC` is not in the date item, the defaultIntegralGains are used
* warning if time difference between event and gain measurement is larger than 5 days

Question: can we create a file with a larger time range?
closed 2018-03-01 by MaxNoe