fact-project/fact-tools #277: Cleanup unused Classes and Processors

MaxNoe, 2017-12-05:
I would like to remove some outdated and unused classes from fact-tools

I identiefied the following candidates:

- [ ] `fact.utils.PullValueFromDistribution`, author @jebuss, not used by any example and documentation unclear, last touch in 2014

- [ ] `fact.auxservice.AuxWebService`, we don't have a web service for this, author @mackaiver

MaxNoe, 2017-12-05:
Please comment @fact-project/fact-tools-core-dev 

Also add stuff, you think we can get rid of

jebuss, 2017-12-05:
>  fact.utils.PullValueFromDistribution, author @jebuss, not used by any example and documentation unclear, last touch in 2014

yes kill it, can't remember why I implemented that

commit 985b7cf6 (2017-12-07): Remove unused classes, fixes #277 (#279)

* Remove unused classes, fixes #277

* Use correct class for floorToQuarterHour

* Remove import of non-existing class
closed 2017-12-07 by MaxNoe