fact-project/fact-tools #289: Events with no found Pedestal Pixel

tarrox, 2017-12-13:
Certain files seem to contain data events, where all pixel are considered shower pixel. This makes the TimeSpread for the Pedestal Pixel fail.

```
Dec 13 2017 15:18:40 [ERROR] stream.runtime.AbstractProcess: Error occurred in process '1': Array of length zero.
Dec 13 2017 15:18:40 [ERROR] stream.runtime.ProcessThread: java.lang.ArrayStoreException: Array of length zero.
        at fact.Utils.arrayMultiplication(Utils.java:418)
        at fact.features.TimeSpread.process(TimeSpread.java:48)
        at stream.runtime.AbstractProcess.process(AbstractProcess.java:126)
        at stream.runtime.AbstractProcess.execute(AbstractProcess.java:184)
        at stream.runtime.ProcessThread.run(ProcessThread.java:129)
```

WIP: Trying to find a file with such an event.

tarrox, 2017-12-13:
Changed the TimeSpread processor to handle the empty pixelset (branch `reinforce_pixelset_processors', by just outputting NaN's in that case, instead of blowing up. Also checked the other processors of the analysis branch whether they could fail, but they seem to be ok with empty pixelsets. Will test the change on the crab1314 sample and create a pull request if everything works.

MaxNoe, 2017-12-13:
These are probably the so called `27s` events

jebuss, 2017-12-13:
I guess so, too. However, I think we should guarantee that the stream does not crash because of them.
I'm not even sure if we want to skip those.

MaxNoe, 2018-01-03:
This is possibly also fixed now in #230

MaxNoe, 2018-01-04:
The PR is merged, can you try again?

closed 2018-02-02 by MaxNoe