fact-project/fact-tools #294: Remove unused, untested processors

MaxNoe, 2017-12-18:
- [ ] `fact.features.source.SourceLineTest` 
  * after an idea by W. Rhode) ;-)
  *  implemented by J. Freiwald in 2013
  *  code not readable, not commented, not tested))

- [ ] `fact.extraction.TimeGradientExtraction` 
  * implemented by F. Temme, not used, not tested

jebuss, 2017-12-18:
I  would vote for keeping those but adding unit tests and maybe find some to evaluate those. 

`fact.features.source.SourceLineTest` is a cleaning method suggested by wolfgang. I think it was a good idea but it was never investigate further due to other priorities at that time. Could be an interesting BA.

`fact.extraction.TimeGradientExtraction` was an Idea for time extraction improvemment based on Sabrinas MA. Same as above.

`fact.extraction.TimeOverThresholdTL` is a nice tool if you want to investigate what is happening on you time series above a given threshold.  However, I agree it would make sense to have an example for this. I think it would not hurt to keep it. Nevertheless, I also don't use it anymore.

MaxNoe, 2017-12-18:
@jebuss The reasons you state would be reasons for me to go forward removing those processors.

The code of `SourceLineTest` is not readable and the actual idea is not mentioned anywhere. It also looks like it doesn't work any more.

MaxNoe, 2017-12-18:
Oh, the `TimeOverThresholdTL` Processor actually head an example, so I keep it. But tests and better docs would be nice to have.

MaxNoe, 2017-12-20:
I added three more Processors (all cleaning)

MaxNoe, 2017-12-20:
@jebuss please comment

commit 811abc5f (2017-12-22): Merge pull request #296 from fact-project/remove_unused

Remove unused processors, fixes #294
closed 2017-12-22 by MaxNoe