fact-project/fact-tools #389: UnixTimeUTC and BoardTime are repeated for 100 times

MaxNoe, 2018-12-17:
When reading zfits files, 100 consecutive events have the same `UnixTimeUTC` and `BoardTime`.

This smells like a bug related to zfits tiles.

MaxNoe, 2018-12-17:
From https://docs.oracle.com/javase/8/docs/api/java/nio/ByteBuffer.html#asShortBuffer

> The content of the new buffer will start at this buffer's current position. Changes to this buffer's content will be visible in the new buffer, and vice versa; the two buffers' position, limit, and mark values will be independent. 

So the buffer does not get advanced.

MaxNoe, 2018-12-17:
fixed in #390

MaxNoe, 2018-12-18:
Now I came to see that this also affected `StartCellData`, so all our DRS calibrations also must have been wrong since that change was made.

MaxNoe, 2018-12-18:
So this should affect all files with `ZTILELEN` > 1

closed 2018-12-18 by MaxNoe

dneise, 2018-12-18:
> Now I came to see that this also affected StartCellData, so all our DRS calibrations also must have been wrong since that change was made.

Was this not realized, since the data in the fits files are already DRS-amplitude calibrated a little bit, and then applying a wrong DRS-calibration is not so easily visible anymore?

But still. Does this mean, the currently running analysis should better be restarted?

MaxNoe, 2018-12-18:
>      Was this not realized, since the data in the fits files are already DRS-amplitude calibrated a little bit, and then applying a wrong DRS-calibration is not so easily visible anymore? 
But still. Does this mean, the currently running analysis should better be restarted?

I jsut checked the `ZTILELEN` of the files at ISDC, only files from 2013 until mid October have `ZTILELEN == 100`, after that, most have `ZTILELEN == 1`, a few have small integer numbers.

This is probably  why it never was realized. Only files before 2013-10-22 are affected heavily. All other files are `ZTILELEN == 1, 2, 3`, so not or only not as badly affected.

commit 5bcff8f4 (2018-12-18): Fix not advancing buffer, fixes #389